pop-os/system76-driver #4: Hidpi daemon notification does not go away on lodpi laptop after removing hidpi monitor

brs17, 2017-12-07:
On a lodpi laptop (like the Gazelle), the hidpi daemon notification will not appear until a hidpi monitor is connected(expected). Once the monitor is disconnected, the notification should go away. It currently does not go away. 

Selecting the notification does not do anything and therefore is not a critical bug.

themccallister, 2018-03-20:
I'm also seeing this on PopOS. The notification for HiDPI never goes away (while I have an external monitor plugged in).

I'm using a Galaga Pro and LoDPI external monitor.

brs17, 2018-03-22:
@themccallister in my experience, clicking on the notification should make it go away.

themccallister, 2018-03-23:
Does not appear to have that affect.

brs17, 2018-03-24:
@djordan2 Have you taken a peak at this recently?

ellisonleao, 2019-11-21:
any updates on this?

sandangel, 2020-03-31:
I have the same issue. I installed PopOS on my macbook pro 2015 retina display

cjohnston1158, 2020-05-14:
Still seeing the same thing on 20.04

sandangel, 2020-05-15:
seem to work fine under wayland

cjohnston1158, 2020-05-18:
I'm not running wayland, so it's still happening for me unfortunately.

Randalix, 2020-06-30:
> I have the same issue. I installed PopOS on my macbook pro 2015 retina display

@sandangel Hi! I'm using Pop Os on macbook pro 2015 retina display as well and having all the same Problems. Maybe we can connect. I'm especially interested in you touchpad settings.
Would be cool if you drop me a line a rpxls1312@gmail.com

sandangel, 2020-06-30:
@Randalix most of the issue I have, like screen tearing, hidpi noti, has been solved by using wayland. I have an install script that help me setup really quick. I will share if you are interested.

Randalix, 2020-07-07:
@sandangel Thanks Man! Highly appreaciated :)