pop-os/system76-driver #6: HiDPI state reverts and gets out of sync when resuming from suspend

cassidyjames, 2017-12-18:
If this is a bug, please use the template below. If this is a question, or a general discussion topic - please post on our subreddit https://reddit.com/r/pop_os - as that is the proper forum for those types of posts.

**Reporter info**

```
NAME="Pop!_OS"
VERSION="17.10 (Artful Aardvark)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 17.10 (Artful Aardvark)"
VERSION_ID="17.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com/"
BUG_REPORT_URL="https://github.com/system76/pop-distro/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=artful
UBUNTU_CODENAME=artful
```

**Issue/Bug Description**

When I'm in LoDPI mode on my Galago Pro (galp2) and resume from suspend, it reverts to HiDPI Mode even though the notification still says it's in LoDPI Mode. Clicking the LoDPI Mode notification doesn't change the scaling, but does dismiss it and trigger the HiDPI Mode notification. _Then_ clicking the notification changes back to LoDPI mode as it should.

**Steps to reproduce (if you know)**

1. Get a galp2 and set to LoDPI Mode
2. Close the laptop to suspend
3. Open the laptop and log back in

**Expected behavior**

The laptop is still in LoDPI mode, and the notifications are still in sync with the scaling state.

**Actual behavior**

The laptop reverts to HiDPI mode, but the notification still incorrectly says it's in LoDPI Mode.

brs17, 2018-03-19:
Is this still the case @cassidyjames ?

cassidyjames, 2018-03-20:
@brs17 Yep.

closed 2025-09-26 by crawfxrd