pop-os/system76-driver #10: HiDPI Daemon sets 1600x900 resolution when external display is attached

jlacroix82, 2017-12-29:
**Reporter info**

```
 * Distribution - (run ```cat /etc/os-release```)
 * Related Application and/or Package Version - system76-driver 17.10.11
```

**Issue/Bug Description**
Note: This is more of a wish/suggestion than a bug.

When you plug in an external display, the driver changes the users resolution to 1600x900. Unfortunately, this is not a good choice for most users.

**Expected behavior**
Instead, when an external display is attached, the resolution should be changed to 1920x1080 with 100% scaling. 1080p matches the majority of external displays, and is the minimum resolution of all external displays sold today. With 1080p, text is still very clear and objects are of a decent size to not cause eyestrain. In addition, 1080p with 100% scaling looks great on this laptop, whereas 1600x900 looks very out of place.

**Other Notes**
1600x900 is a great resolution for people with vision handicaps, so maybe it could be an option as some sort of accessibility mode for those people, if it's not already?

cassidyjames, 2018-01-05:
@jlacroix82 what hardware are you on? If you're on something like a 13"  3200×1800 display, the daemon is doing the correct thing: it's keeping everything the exact same _physical_ size while halving the resolution to trigger loDPI. 

1080p at 13" is too tiny for many users (and is why we don't offer a 1080p or 4K—effectively 1080p pixel-doubled—display on the Galago Pro, but chose 3200×1800). This also offers the most pixel clarity (due to pixels being aligned); instead of trying to draw one virtual pixel over 1.67 physical pixels (which is impossible without introducing blurriness), it's drawing them over two physical pixels. And lastly, it keeps the window layout and virtual real estate the same regardless of being in LoDPI or HiDPI mode.

I wrote a series of articles about this leading up to the Galago Pro release that helps cover this:
- [What is HiDPI (and why does it matter)?](https://medium.com/elementaryos/what-is-hidpi-and-why-does-it-matter-b024eabea20d)
- [Top 3 Misconceptions About HiDPI](https://medium.com/elementaryos/top-3-misconceptions-about-hidpi-f5ef493d7bf8)
- [HiDPI is More Important than 4K](https://medium.com/elementaryos/hidpi-is-more-important-than-4k-b1b9bb110b43)

Of course, you are welcome to disable the daemon and/or manually manage your display resolution, but we chose the defaults to be the best for the majority of people for very good reasons. :wink:

djordan2, 2018-03-14:
This should be fixed as of 17.10.23.

closed 2018-03-19 by brs17