pop-os/system76-driver #9: HiDPI Daemon Puts All Applications on a single Workspace

jlacroix82, 2017-12-29:
**Reporter info**

```
 * Distribution - Ubuntu 17.10 64-bit
 * Related Application and/or Package Version - system76-driver 17.10.11
```

**Issue/Bug Description**
Normally, when you are working with multiple displays, GNOME will move applications back to the monitor they were on when you reattach a display. With the HiDPI driver, all applications are placed on a single workspace, forcing the user to manually drag all the applications manually back to the desired workspace.

**Steps to reproduce (if you know)**
1.) Attach an external display
2.) Open several applications, but them on different workspaces
3.) Remove the external display, then reattach it
4.) All the applications you had open are now on a single workspace

**Expected behavior**
By default, GNOME will place applications back to the workspace they were on before disconnecting the display. The HiDPI driver circumvents this GNOME feature and forces users to do this manually.

jlacroix82, 2018-02-15:
In speaking with Jeremy and working with him on reproducing this, we determined that this is a GNOME issue and is triggered when switching monitor configurations while also having the option enabled that utilizes workspaces only on the main display. This is a consequence of having to redo my monitor configuration manually due to the HiDPI daemon not remembering settings, but the underlying issue is GNOME-related and not specific to the HiDPI daemon.

Therefore, we can either close this, or keep it open to track this issue. Please let me know your thoughts.

djordan2, 2018-03-14:
This is a GNOME issue, but improvements to the HiDPI daemon reduce the need for manual configuration.

brs17, 2018-03-19:
@jlacroix82 So @djordan2 did what he could to make the HiDPI daemon work as nicely as it can with upstream GNOME. The appropriate thing would definitely be to file a bug upstream. Would you like to do that or shall I?

jlacroix82, 2018-03-20:
> @jlacroix82 So @djordan2 did what he could to make the HiDPI daemon work as nicely as it can with upstream GNOME. The appropriate thing would definitely be to file a bug upstream. Would you like to do that or shall I?

I thought there was already a bug for this, I'm not able to find it right now but I'll look again.

closed 2025-09-26 by crawfxrd