pop-os/system76-driver #20: Cannot install Firmware updates on i3 WM

Mange, 2018-02-13:
**Reporter info**

```
 › cat /etc/os-release
NAME="Ubuntu"
VERSION="17.10 (Artful Aardvark)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 17.10"
VERSION_ID="17.10"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=artful
UBUNTU_CODENAME=artful

 › apt policy system76-driver
system76-driver:
  Installed: 17.10.18
  Candidate: 17.10.18
  Version table:
 *** 17.10.18 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu artful/main amd64 Packages
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu artful/main i386 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description**

Trying to install firmware does not work and results in a backtrace from Python. The service running in the background crashes on startup with the same error.

```
 › sudo system76-firmware
2018-02-13 07:57:14,801  INFO  Verified manifest signature...
2018-02-13 07:57:14,802  INFO  Fetching bd6f74a479ebaed9d83aac2a9d3149b8410aa57ec70a22a53e5b25bd92c0846bb2708085d76fabb3893ac5fe126b73a1 with cache /var/cache/system76-firmware
2018-02-13 07:57:14,804  INFO  Fetching b5a329d651087b0027a58bda120317cc28102ebcda888c41c2f421e1d46336f422532af10b02a7602ca4420057696994 with cache /var/cache/system76-firmware
2018-02-13 07:57:14,838  INFO  Fetching 510f23b8dcf1d5f324ae67a9a1dd99e24ef0e0eb815b1e14d7bc810101383f689f0f8169fbce48ebbc77c3c1a84cedd8 with cache /var/cache/system76-firmware
2018-02-13 07:57:16,030  INFO  Verified manifest signature...
2018-02-13 07:57:16,030  INFO  Fetching bd6f74a479ebaed9d83aac2a9d3149b8410aa57ec70a22a53e5b25bd92c0846bb2708085d76fabb3893ac5fe126b73a1 with cache /var/cache/system76-firmware
2018-02-13 07:57:16,032  INFO  Fetching b5a329d651087b0027a58bda120317cc28102ebcda888c41c2f421e1d46336f422532af10b02a7602ca4420057696994 with cache /var/cache/system76-firmware
2018-02-13 07:57:16,063  INFO  Fetching 510f23b8dcf1d5f324ae67a9a1dd99e24ef0e0eb815b1e14d7bc810101383f689f0f8169fbce48ebbc77c3c1a84cedd8 with cache /var/cache/system76-firmware
2018-02-13 07:57:16,231  ERROR  Error calling _run_firmware_updater()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/system76driver/firmware.py", line 722, in run_firmware_updater
    ret = _run_firmware_updater(reinstall, notification)
  File "/usr/lib/python3/dist-packages/system76driver/firmware.py", line 688, in _run_firmware_updater
    if confirm_dialog(data) == 76:
  File "/usr/lib/python3/dist-packages/system76driver/firmware.py", line 482, in confirm_dialog
    user_name, display_name, environ = get_user_session()
  File "/usr/lib/python3/dist-packages/system76driver/firmware.py", line 452, in get_user_session
    user_session_pids = subprocess.check_output(['pgrep', '-P', user_pid]
  File "/usr/lib/python3.6/subprocess.py", line 336, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 418, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['pgrep', '-P', '2524']' returned non-zero exit status 1.
```

**Steps to reproduce (if you know)**

1. Use i3 WM as your window manager.
2. Open a terminal.
3. Run `sudo system76-firmware`

**Expected behavior**

1. Installation proceeds according to plan.

**Other Notes**

Stepping through the code it looks like it is trying to find the first child process of the WM to get my current `env`. This seems to be a problem with i3 as it has no children.

```
› who -u
mange    tty7         2018-02-08 09:03  old         2524 (:0)
› pgrep -P 2524; echo $?
1
› pstree -s -p 2524
systemd(1)───lightdm(1551)───lightdm(1989)───i3(2524)
```

My desktop processes seem to live straight under `systemd`.

```
 › pgrep gnome-terminal | xargs pstree -s -p
systemd(1)───systemd(2512)───gnome-terminal-(8622)─┬─zsh(3024)
                                                   ├─zsh(11869)
                                                   ├─zsh(14904)───nvim(9476)─┬─python3(10472)
                                                   │                         └─{nvim}(9478)
                                                   ├─zsh(15107)
                                                   ├─zsh(15639)
                                                   ├─zsh(27585)───xargs(18287)───pstree(18288)
                                                   ├─{gnome-terminal-}(8623)
                                                   ├─{gnome-terminal-}(8624)
                                                   └─{gnome-terminal-}(8625)
```

Shalzz, 2018-02-14:
Happens for me even on a gnome system
```
2018-02-14 14:44:33,218  ERROR  Error calling _run_firmware_updater()
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/system76driver/firmware.py", line 722, in run_firmware_updater
    ret = _run_firmware_updater(reinstall, notification)
  File "/usr/lib/python3.6/site-packages/system76driver/firmware.py", line 688, in _run_firmware_updater
    if confirm_dialog(data) == 76:
  File "/usr/lib/python3.6/site-packages/system76driver/firmware.py", line 482, in confirm_dialog
    user_name, display_name, environ = get_user_session()
  File "/usr/lib/python3.6/site-packages/system76driver/firmware.py", line 452, in get_user_session
    user_session_pids = subprocess.check_output(['pgrep', '-P', user_pid]
  File "/usr/lib/python3.6/subprocess.py", line 336, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 418, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['pgrep', '-P', '']' returned non-zero exit status 2.
```

brs17, 2018-03-19:
@Mange @shalzz thank you for filing this issue, is the problem still occurring on your machines? 

@jackpot51 Do you have any idea as to why this is happening?

I would recommend you both opening support tickets since this is a customer specific issue.

Mange, 2018-03-20:
```bash
sudo apt update
sudo apt dist-upgrade
sudo system76-driver
```

Everything seems to work now! Thanks!

closed 2018-03-20 by Mange