pop-os/system76-driver #25: system76-driver creating endless login loop

flyinprogrammer, 2018-03-15:
**Reporter info**

```
 * Distribution - (run ```cat /etc/os-release```)
$ cat /etc/os-release
NAME="Pop!_OS"
VERSION="17.10 (Artful Aardvark)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 17.10 (Artful Aardvark)"
VERSION_ID="17.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com/"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=artful
UBUNTU_CODENAME=artful

 * Related Application and/or Package Version - apt policy $PACKAGE NAME
```
$ apt policy system76-driver
system76-driver:
  Installed: (none)
  Candidate: 17.10.24
  Version table:
     17.10.24 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu artful/main amd64 Packages
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu artful/main i386 Packages
```

**Issue/Bug Description**
When system76-driver is installed I can no longer login to my laptop. Instead I login, the screen flashes, and then I'm booted directly back to the login screen. I suspect it has something to do with the HiDPI bit of the driver because if I login with a secondary monitor then the login succeeds. I also suspect that it's the driver because upon removing it I was able to log back into my computer.

**Steps to reproduce (if you know)**
galp2+latest packages. turn on your machine, go to login, it will kick you back out to the login screen.

**Expected behavior**

I should be able to have the  driver installed and login to the host.

**Other Notes**

Happy to get you any logs you want, just wasn't sure what would be helpful, syslog didn't seem terribly helpful. Also more than happy to film this issue and upload it to YouTube for proof.

jackpot51, 2018-03-15:
Thanks for submitting the issue. I have forwarded the information to @djordan2 - I hope we can get this fixed soon!

late-latte, 2018-04-02:
@jackpot51  Hi, have this been fixed yet? I also encountered similar problem with oryp+ess, although I'm using Arch + nvidia-dkms + minimal gnome + lightdm. The problem seems to lie on lightdm (or perhaps display managers in general), where it spouted g object or glib schemas not found. No matter what I do with /etc/environment, or how I regenerated /usr/share/glib-2.0/schemas, it's still stuck being unable to access desktop. Common cli login works fine.

OOT: Can you guys modularize DAC fix from the general system76-driver? I only need that fix (assuming it's not fixed out of the box), since HiDPI, brightness, keyboard light, and GPU works just fine by minimal tinkering with Arch, and system76-driver forced me to use grub while I'm using systemd-boot just fine. I'm not entirely sure how to check/access/manipulate DAC from terminal.

closed 2020-10-25 by flyinprogrammer