pop-os/system76-driver #43: system76-driver 18.04.23 depends on gnome-shell, while 18.04.22 did not

dtfinch, 2018-06-08:
**Distribution (run `cat /etc/os-release`):**

    NAME="Ubuntu"  
    VERSION="18.04 LTS (Bionic Beaver)"  
    ID=ubuntu  
    ID_LIKE=debian  
    PRETTY_NAME="Ubuntu 18.04 LTS"  
    VERSION_ID="18.04"  
    HOME_URL="https://www.ubuntu.com/"  
    SUPPORT_URL="https://help.ubuntu.com/"  
    BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"  
    PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"  
    VERSION_CODENAME=bionic  
    UBUNTU_CODENAME=bionic

**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

system76-driver

**Issue/Bug Description:**

Version 18.04.23 created a dependency on gnome-shell, via a new dependency on gnome-shell-extension-system76-power, which 18.04.22 did not have.

When I try to upgrade system76-driver, apt-get wants to install 47 additional packages (I use XFCE):

    The following NEW packages will be installed:
      bolt evolution-data-server evolution-data-server-common
      gir1.2-accountsservice-1.0 gir1.2-gck-1 gir1.2-gcr-3 gir1.2-gdm-1.0
      gir1.2-geoclue-2.0 gir1.2-gnomebluetooth-1.0 gir1.2-gweather-3.0
      gir1.2-mutter-2 gir1.2-rsvg-2.0 gir1.2-upowerglib-1.0 gjs
      gnome-control-center gnome-control-center-faces gnome-online-accounts
      gnome-shell gnome-shell-extension-system76-power libcamel-1.2-61
      libcolord-gtk1 libebackend-1.2-10 libebook-1.2-19 libebook-contacts-1.2-2
      libecal-1.2-19 libedata-book-1.2-25 libedata-cal-1.2-28
      libedataserver-1.2-23 libgdm1 libgjs0g libgoa-backend-1.0-1 libmozjs-52-0
      libmutter-2-0 libnss-myhostname libphonenumber7 libxcb-res0 mutter
      python3-macaroonbakery python3-nacl python3-protobuf python3-pymacaroons
      python3-rfc3339 python3-tz system76-power ubuntu-session xserver-xorg-legacy
      xwayland
    The following packages will be upgraded:
      system76-driver
    1 upgraded, 47 newly installed, 0 to remove and 0 not upgraded.

**Steps to reproduce (if you know):**

    apt-get update
    apt-get dist-upgrade

**Expected behavior:**

I expected a stable update without drastic changes since 18.04 is an LTS release.

**Other Notes:**

My desktop runs XFCE (via xubuntu-desktop) rather than Gnome. Some of the packages it wants to install are things I've tried to keep off of my system.

Maybe the gnome-shell-extension-system76-power dependency could be changed from Depends to Recommends? Apt would still install it by default, but it'd no longer be mandatory.

djordan2, 2018-06-08:
We're updating the gnome-shell-extension-system76-power momentarily, so that the extension will not depend on gnome-shell.  The intent here was to ensure that system76-power and the only current graphical interface to it get installed, but one important bit of the dependency shuffling wasn't released with the others.

djordan2, 2018-06-08:
I think this should be resolved now.  Gnome Shell is no longer depended on by the system76-power related packages.  Thanks for reporting this, and helping us catch this quickly!

closed 2018-06-08 by djordan2