pop-os/system76-driver #52: Python error when upgrading package

paul, 2018-08-01:
**Distribution (run `cat /etc/os-release`):**
 
```
NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
$ apt policy system76-driver
system76-driver:
  Installed: 18.04.26~1532726241~18.04~13ce8b6
  Candidate: 18.04.26~1532726241~18.04~13ce8b6
  Version table:
 *** 18.04.26~1532726241~18.04~13ce8b6 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main i386 Packages
        100 /var/lib/dpkg/status
                                  
```

**Issue/Bug Description:**

When I did an `apt update && apt upgrade` this morning, I got some error output:

```
Setting up system76-driver (18.04.26~1532726241~18.04~13ce8b6) ...
INFO	product_version: 'oryp3-ess'
ERROR	Error running actions:
Traceback (most recent call last):
  File "/usr/bin/system76-driver-cli", line 71, in <module>
    action_runner = ActionRunner(product['drivers'])
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 235, in __init__
    if action.isneeded:
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 161, in isneeded
    self._isneeded = self.get_isneeded()
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 406, in get_isneeded
    params = set(current.split())
AttributeError: 'list' object has no attribute 'split'
Setting up pop-shop (0.2.9~1532969963~18.04~d2fb21b) ...
```

The upgrade seemed to finish ok, but I haven't rebooted yet.

**Steps to reproduce (if you know):**

I get the same error when running `system76-driver-cli`:

```
$ sudo system76-driver-cli
INFO	product_version: 'oryp3-ess'
ERROR	Error running actions:
Traceback (most recent call last):
  File "/usr/bin/system76-driver-cli", line 71, in <module>
    action_runner = ActionRunner(product['drivers'])
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 235, in __init__
    if action.isneeded:
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 161, in isneeded
    self._isneeded = self.get_isneeded()
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 406, in get_isneeded
    params = set(current.split())
AttributeError: 'list' object has no attribute 'split'
```

**Expected behavior:**

No errors in the `apt upgrade` output.

**Other Notes:**

paul, 2018-08-01:
I don't know much python, but it looks like the error happens when its parsing the output of `/etc/kernelstub/configuration`, so here's that, too:

```
{
  "default": {
    "kernel_options": [
      "quiet",
      "loglevel=0",
      "vga=current",
      "systemd.show_status=false"
    ],
    "esp_path": "/boot/efi",
    "setup_loader": true,
    "manage_mode": true,
    "force_update": false,
    "live_mode": true,
    "config_rev": 3
  },
  "user": {
    "kernel_options": [
      "quiet",
      "loglevel=0",
      "vga=current",
      "systemd.show_status=false",
      "splash"
    ],
    "esp_path": "/boot/efi",
    "setup_loader": true,
    "manage_mode": true,
    "force_update": false,
    "live_mode": false,
    "config_rev": 3
  }
}
```

brs17, 2018-08-01:
@paul Thank you for filing this issue. 

This looks like it may be a regression from this commit: https://github.com/pop-os/kernelstub/commit/8865fb5e03f6336f1adf2777dacf9576668c4d5e . I have been able to reproduce this issue in the lab, should be relatively close to finding the fix.

commit 42649cb9 (2018-08-01): Don't split() on a list, fix #52 (#53)
closed 2018-08-01 by mmstick