pop-os/system76-driver #54: oryx-pro2 system76-driver upgrade error

pcnate, 2018-08-03:
**Distribution (run `cat /etc/os-release`):**
 ```
NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
pcnate@pop-os:~$ apt policy system76-driver
system76-driver:
  Installed: 18.04.26~1532726241~18.04~13ce8b6
  Candidate: 18.04.26~1532726241~18.04~13ce8b6
  Version table:
 *** 18.04.26~1532726241~18.04~13ce8b6 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main i386 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**
when upgrading via command line with `sudo apt-get dist-upgrade`

```
Setting up apt-config-icons-large (0.12.0-3) ...
Setting up system76-driver (18.04.26~1532726241~18.04~13ce8b6) ...
INFO	product_version: 'oryp2'
ERROR	Error running actions:
Traceback (most recent call last):
  File "/usr/bin/system76-driver-cli", line 71, in <module>
    action_runner = ActionRunner(product['drivers'])
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 235, in __init__
    if action.isneeded:
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 161, in isneeded
    self._isneeded = self.get_isneeded()
  File "/usr/lib/python3/dist-packages/system76driver/actions.py", line 406, in get_isneeded
    params = set(current.split())
AttributeError: 'list' object has no attribute 'split'
Setting up apt-config-icons-hidpi (0.12.0-3) ...
```
https://github.com/pop-os/system76-driver/blob/42649cb9e497e09bf2b3448efebe96fc925c5524/system76driver/actions.py#L235
https://github.com/pop-os/system76-driver/blob/42649cb9e497e09bf2b3448efebe96fc925c5524/system76driver/actions.py#L161
This one is different on my system
https://github.com/pop-os/system76-driver/blob/42649cb9e497e09bf2b3448efebe96fc925c5524/system76driver/actions.py#L406

**Steps to reproduce (if you know):**

**Expected behavior:**

**Other Notes:**

complete log
https://gist.github.com/pcnate/e943526a56b8d38f86814822cb367372

pcnate, 2018-08-03:
It does not show up when trying to upgrade again. How can I be sure it is installed properly and not going to borke my system on the next reboot?

brs17, 2018-08-03:
We got an issue about this traceback https://github.com/pop-os/system76-driver/issues/52 and pushed a fix https://github.com/pop-os/system76-driver/pull/53.

So all should be well now (after updating to the latest version of the system76-driver)

pcnate, 2018-08-03:
I'll have to give it a try when I get home. Unable to SSH into it currently

pcnate, 2018-08-07:
Haven't had any new version pull.

```
pcnate@pop-os:~/$ apt policy system76-driver
system76-driver:
  Installed: 18.04.26~1532726241~18.04~13ce8b6
  Candidate: 18.04.26~1532726241~18.04~13ce8b6
  Version table:
 *** 18.04.26~1532726241~18.04~13ce8b6 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main i386 Packages
        100 /var/lib/dpkg/status
```

https://gist.github.com/pcnate/e943526a56b8d38f86814822cb367372#file-next-update

brs17, 2018-08-08:
Sorry for the delay, the patch should have just been released now (I see the update come up on my machine).

closed 2018-08-08 by brs17