pop-os/system76-driver #59: Oryx4-b Backlight Control Issue/DPCD PWM Reset

danieljohnson2, 2018-08-29:
**Distribution (run `cat /etc/os-release`):**
NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.1 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
  
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-driver:
  Installed: 18.04.27~1533680343~18.04~e74d1c2~dev
  Candidate: 18.04.27~1533680343~18.04~e74d1c2~dev
  Version table:
 *** 18.04.27~1533680343~18.04~e74d1c2~dev 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic/main amd64 Packages
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic/main i386 Packages
        100 /var/lib/dpkg/status

But I have installed Pop-OS, just to test this, and it repros exactly the same. I normally run with gnome-session, but I have tried without and that also does not help.

**Issue/Bug Description:**
On Oryx4-b (only) the System76 driver adjusts a mode byte (function apply_dpcd_pwm_fix in daemon.py). It does this repeatedly, in case it resets itself.

But this driver stops trying to do this after 60 seconds. After that, if the mode resets the backlight goes to max brightness, and all brightness controls (except xrandr --brightness, oddly) cease to have any effect.

There are many things that can cause this mode to reset.

**Steps to reproduce (if you know):**
You need to wait at least 60 seconds after logging in before attempting to repro this.

After that, you can put the computer to sleep and awaken it, and that will usually trigger this. Changing screen resolutions can also trigger it, but I find it sometimes takes several tries for this to work.

The Linux port of Doom 3 (package dhewm3) triggers this issue for more almost every time I start it; I note that I play this game at a lower-than-native-resolution due to the game's aspect ratio issue. Also note that the game actually has to start, so you do need the Doom 3 data files installed to repro this way. But if you can play Doom 3 like this, it's the most reliable repro. Even this is not 100% guaranteed.

**Expected behavior:**
Ideally, the Oryx4 should not go to max brightness unless commanded by the user, and the brightness controls should all always affect the screen brightness.

Looking at the code, my expectations are a bit lower- events that trigger this problem will cause the laptop to go to max brightness for at most 3 seconds, then resume normal function. Brightness controls should work again after normal service is resumed.

**Other Notes:**
I've been working with System76 support on this issue.

You can restore proper backlight function temporarily by restarting the System76 driver, via 'systemctl restart system76-driver.service'. After 60 seconds, you will again be able repro.

I found a workaround for this; the relevant code for this is installed at /usr/lib/python3/dist-packages/system76driver/daemon.py. Edit this file and find line 627, which is "if nhash != chash:"; change this to "if True:" instead (preserve the indentation though, because Python). This defeats the check on the login.sessions (whatever they are) and keeps restarting the 60-second loop forever.

Now, if the mode ever changes, it will be reset within 3 seconds or so. Of course, this is terribly poll-tastic, and will waste a tiny bit of CPU while it keeps Python 3 resident in memory forever. This is still much better than a busted backlight.

jackpot51, 2018-09-06:
Thanks for the information, I believe we can come up with a solution quickly

brs17, 2018-09-06:
This definitely sounds like a bug with very hacky fix we did to get backlight working. :'(

danieljohnson2, 2018-09-06:
The safest thing may be just to keep stomping that byte every 3 seconds forever; no need to anticipate what will set off the 'raaaar must microwave user's face' thing.

I've been running my hacked up driver and it doesn't seem to have done anything to the battery life, so far as I can tell.

jackpot51, 2018-09-09:
We will release a driver update soon to fix this

danieljohnson2, 2018-09-09:
I'm glad to hear that. I'll try it as soon as I can get my hands on it, and I'll let you know if it worked for me.

brs17, 2018-09-10:
@danieljohnson2 So I just attempted to reproduce your issue.

This is what I did and the result I got. Is this the bug you were describing?

1. Boot machine
2. Change screen brightness
3. Let it sit for 60 seconds
4. Suspend the machine.
5. Resume machine.
After resume the screen brightness is at 100% and cannot be changed.

danieljohnson2, 2018-09-10:
Yes, pretty nearly.

I find that the bug does not require that I change the screen brightness initially. However, reducing the brightness to a low level will make it much more obvious when it happens. The 60 seconds wait is the trick; the daemon has to get to the state where it's waiting for login.sessions to change.

I suppose that is why your Dvorak enthusiasts need to sleep and resume *twice*; this is how they spend 60 seconds. But I am definitely running on Qwerty and still experience the issue, and see no reason why the keyboard layout would have any effect.

They say that logging out resets it. I've just tried that, and it worked: logging out it by itself enough to restore the backlight intensity, even without logging in again. I am guessing that this is what the login.session check is detecting.

One of the Dvorakites says that anything that might blank the screen triggers it. I've observed this too, though not consistently- but occasionally if the display is turned off according to the power-saving options, though the machine is still awake, the backlight goes feral.

My Oryx Pro is also a 17 inch model. Is that what the 'b' in 'oryx4-b' stands for?

Anyway, I think it's very likely that issue [#344](https://github.com/pop-os/pop/issues/344) and this one are the same issue.

brs17, 2018-09-11:
Sounds like we are on the same page. You are right, I did not need to change the brightness, but it does make it more obvious.

Yes, we need a way to differentiate the 15" and 17" models so we add the 'b'.

mmstick, 2018-09-11:
> This defeats the check on the login.sessions (whatever they are) and keeps restarting the 60-second loop forever.

There's a session change every time you switch TTYs, log out, log in, etc. That's what that is looking for. We found that each time the session would change, the NVIDIA driver would reset that bit whereas the Intel driver would not, but I suppose there's something else that's also triggering the driver to overwrite it. I don't think it'd be too costly to just do it forever. May even have less cost than the current approach.

brs17, 2018-09-11:
@mmstick: related, and I figured I would put it here as a reminder: currently screen brightness does not work at all on autologin users (such as the oem mode)

danieljohnson2, 2018-09-12:
@mmstick, perhaps I should mention that whatever is going on, it happens just the same whether in Intel or NVIDIA mode. Can the NVIDIA driver reset the magic bit when it is turned off?

m160, 2018-09-14:
The latest update seem to have fixed it for me. Thanks guys :)

danieljohnson2, 2018-09-14:
The latest update did not help. It reset daemon.py back to stock, reversing out my hackjob workaround, and restarted the driver.

Restarting the driver gets it working again for 60 seconds, but after that it once again thinks it should be the one to fill my dark soul with light.

danieljohnson2, 2018-09-20:
Can we get an ETA on this issue?

m160, 2018-09-26:
@jackpot51 - Any update on the fix?

m160, 2018-10-02:
@danieljohnson2 - I found a little workaround. Press ctrl + alt + F1, that fixes it for me until next time the display turns off. A bit more convenient than rebooting or logging out completely.

jackpot51, 2018-10-03:
Sorry about the delay. I am still weighing the options in terms of a more permanent firmware fix that would also work when the driver is not installed

danieljohnson2, 2018-10-03:
That does sound like a better fix, if it is possible, but it's taking quite a while. I find it irritating to have to re-apply my hack every time I get a System76 driver update that re-installs the daemon, and I suspect you have quite a few customers who aren't comfortable tweaking daemon.py like I am doing.

If the hacky fix is easy to do, perhaps it's worth doing just as a stopgap, even if you will eventually do a better firmware-based fix.

jackpot51, 2018-10-04:
If you have a modification that works in the mean time, would you mind making a PR?

danieljohnson2, 2018-10-04:
I have just attempted to provide a pull request for this (#66). This is baby's first pull request, so please be gentle!

jackpot51, 2018-10-05:
Thanks very much @danieljohnson2 ! If it needs some polish, I will make changes to it

danieljohnson2, 2018-10-05:
By all means, make it your own.

I think I may have violated the System76 coding standards: I put in comments.

danieljohnson2, 2018-10-24:
19 days and nothing. Perhaps I shouldn't have been all snarky about comments...

m160, 2018-10-27:
I clean installed 18.10 in hopes that it would fix it, but the issue still remains. Closing in on two months since it was reported now. Did I make a mistake buying a System76 laptop?

danieljohnson2, 2018-10-27:
@m160, maybe you did- they've not been super-quick about this problem.

Anyway, I've upgraded my Oryx4-b to 18.10, which was a bit troublesome, but my workaround still works just as before. See "Other Notes" in the original post above if you want to try it.

I'm trying to get them to ship a nicer version of this workaround (pull #66) as part of the driver, but it seems to take a certain amount of nagging.

danieljohnson2, 2018-10-27:
On a Saturday no less! Does this show @jackpot51's dedication to customer service, or the irresistible power of whining?

Either way, thank you.

The fix is now available from your PPA, I've updated, and it works for me (on 18.10 now)..

m160, 2018-10-27:
One does not simply suggest that having bought a System76 laptop might've been a mistake :P
It works here too.

I have a quick question about the NVIDIA graphics mode if I may, the fans push so much air all the time, even when the computer is idle. Is this normal? With Fn + 1 I can choose between two settings, loud and LOUD. Shouldn't the fans quite down when idle at least?

closed 2018-10-28 by jackpot51