pop-os/system76-driver #61: `xbacklight` cannot control brightness

ExpandingMan, 2018-09-05:
# System info (`inxi -SMGxxxz`)
```
System:    Host: oryx-1 Kernel: 4.15.0-33-generic x86_64 bits: 64 gcc: 7.3.0
           Desktop: Gnome 3.28.3 (Gtk 3.22.30-1ubuntu1) info: gnome-shell dm: gdm3 Distro: Pop!_OS 18.04 LTS
Machine:   Device: laptop System: System76 product: Oryx Pro v: oryp4 serial: N/A  Chassis: type: 10 serial: N/A
           Mobo: System76 model: Oryx Pro v: oryp4 serial: N/A
           UEFI: American Megatrends v: 7.006S76 date: 05/18/2018
Graphics:  Card: Intel Device 3e9b bus-ID: 00:02.0 chip-ID: 8086:3e9b
           Display Server: x11 (X.Org 1.19.6 ) drivers: modesetting,nvidia (unloaded: fbdev,vesa,nouveau)
           Resolution: 1920x1080@59.96hz
           OpenGL: renderer: Mesa DRI Intel HD Graphics (Coffeelake 3x8 GT2)
           version: 4.5 Mesa 18.0.5 (compat-v: 3.0) Direct Render: Yes
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
`xbacklight`

**Issue/Bug Description:**
Setting the screen brightness with `xbacklight` has no effect.  Keyboard brightness controls within gnome seem to work normally, but I don't know how it acheives that.  I seem to recall this was also the same when I had `nvidia` active, but honestly I haven't double checked.

**Steps to reproduce (if you know):**
`xbacklight -inc 20` or any other `xbacklight` command.

**Expected behavior:**
Backlight intensity changes.

jackpot51, 2018-09-06:
GNOME uses /sys/class/backlight to set brightness. Please let me know if you can utilize that interface instead

ExpandingMan, 2018-09-06:
Yes, this works thanks.  Is it expected behavior that `xbacklight` does not work?  From what I can tell, with the Intel graphics driver it should always work in X.  My supposition was that the reason it was not working might have something to do with `system76-driver`.

If this is expected behavior we can of course close this issue.

jackpot51, 2018-09-08:
This is expected behavior, for xbacklight to not operate, and is a result of a GNOME decision. It is not related to anything System76 has done.

ExpandingMan, 2018-09-08:
`xbacklight` is not working even *outside* of GNOME in `i3`.

Anyway, if you're sure there's no way that `system76-driver` can cause this, please feel free to close the issue.  `xbacklight` supposedly would work in X with the Intel Mesa driver, so this was next on my list of things that could stop it from working, but I of course was never sure it had anything to do with this.

Thanks.

closed 2018-09-09 by jackpot51