pop-os/system76-driver #72: TouchPad Not Working

rayush7, 2018-10-26:
Hi guys,

I recently installed PopOS 18.04 on my laptop but the touchpad is not working. I installed the drivers specified on the System76 website but still the touchpad doesn't work at all. Actually the touchpad is not even detected when I use the command xinput on the terminal. I am able to work using an optical mouse connected to the computer through a usb cable but please help me resolve the issue. Earlier I had windows in my laptop and touchpad was working well in it. Can anyone help me with this?

Thank you

brs17, 2018-10-26:
@rayush7 what model laptop do you have?

rayush7, 2018-10-26:
@brs17 
I got my laptop assembled from [PC Specialist](https://www.pcspecialist.co.uk/)
The specs of the laptop are mentioned [here](https://www.pcspecialist.co.uk/notebooks/vyper-15/)

Is there anything I can do to resolve the issue?

jackpot51, 2018-10-26:
> I installed the drivers specified on the System76 website

What do you mean by that? What drivers were specified?

rayush7, 2018-10-26:
@jackpot51 I mean I followed the procedure mentioned [here](https://support.system76.com/articles/system76-driver/) to install the normal drivers and the nvidia driver. But it didn't help much.

freechelmi, 2022-07-20:
system76-driver project is not meant to support other brand computers.

You can open the same issue on our fork here https://github.com/ekimiateam/linux-universal-driver/issues

Where we add support for all other computers when possible.

crawfxrd, 2022-07-22:
Not System76 hardware.

closed 2022-07-22 by crawfxrd