pop-os/system76-driver #110: Documentation request

drNoob13, 2019-06-04:
**Hardware: **
Few System76 Oryx pro 4 laptops with GTX 1070 Max-Q that run on Ubuntu 16.04 LTS.

**Distribution (run `cat /etc/os-release`):**
```bash
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="16.04.6 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.6 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
``` 

 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
Not compatible on Ubuntu 16.04.

**Issue/Bug Description:**
There is no information for System76 laptop owners to build the drivers.

**Steps to reproduce (if you know):**
There is no documentation for System76 customers to build the driver packages for their System76 laptops. System76 doesn't provide ppa for driver installation in Ubuntu 16.04.

**Expected behavior:**
Could you provide build instruction/documentation so that System76 laptop owner can build the driver by themselves? Thank you.

**Other Notes:**

drNoob13, 2019-06-27:
Hello,

Could someone from System76 please answer the question?

brs17, 2019-06-27:
@drNoob13 ,

The Oryx Pro 4 was released in tandom with 18.04 where there was a significant amount of work put in to make sure the machine worked to our standards. Therefore 16.04 is not officially supported. That being said if you contact System76 support (https://support.system76.com/) they can potentially provide you information for what things people have gotten to work (and what doesn't) in 16.04.

closed 2019-06-27 by brs17

drNoob13, 2019-06-28:
@brs17 : 

Thanks for your response. I understand and respect your company's priority and focus. Actually, I was not asking for Ubuntu 16.04 support. What was asking is how to compile the package you provide here. A Readme for your support version on this repo, be it either PopOS or Ubuntu 18.04--the OSes you are supporting, is greatly welcome. I will figure out my own how to adapt it to our need--Ubuntu 16.04. Again, I don't ask for your technical support on the target platform you are not supporting.

I believe producing good documentation will help your company in the long run. Not all customers use PopOS or Ubuntu 18.04. I'm an enterprise user who would like to support S76 (and we bought a few S76 oryx pro laptops already under either my name or my company name). However, with a lack of documentation on compiling your basic packages like firmware, drivers, etc., it'll be hard for me to keep recommending S76 to my colleagues.

I appreciate your reconsideration on this matter.

Best regards

wingunder, 2019-09-17:
> Actually, I was not asking for Ubuntu 16.04 support. What was asking is how to compile the package you provide here.

You can compile the .deb packages with the following command:
`VENDORED=0 dpkg-buildpackage -b`

I have an OryxPro on Debian (testing) and it worked for me.