pop-os/system76-driver #168: No Audio output from my speakers

jonneymendoza, 2020-11-09:
**Distribution (run `cat /etc/os-release`):**
 
 ```NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
Chrome youtube, spotify

**Issue/Bug Description:**

Hi i am getting no sound at all from my speakers that are connected to a line-out connector on my desktop computer. Same setup works fine on when i dual boot to windows 10.

Here is my hardware specs:

AMD Ryzen Threadripper 3960X 24-Core Processor128gb

ddr4 ram

Asus ROG ZENITH 2 EXTREME Motherboard

M.2 4Tb SSD DRIVE.

NVIDIA 2080TI Gpu

Pop OS 20.10

ASUS PA32UC-K 32 Inch Professional Monitor

The audio only works through the headphone jack of my desktop or through my 4k monitor display connected via displayPort.

I tried following the guide/troubleshoot here https://support.system76.com/articles/audio/ and installed pulseAudio volume controler and tried to manually select the line-out output and still no sound is coming out of it

**Steps to reproduce (if you know):**
Play any media with sound output  from chrome or spotify or any other program that outputs sound

**Expected behavior:**
Play the sound through my line-out connected speakers

**Other Notes:**
Works if i set the audio output to my 4k monitor speakers or the headphone jack connected to my computer destop

jacobgkau, 2020-11-09:
> Hi i am getting no sound at all from my speakers that are connected to a line-out connector on my desktop computer. Same setup works fine on when i dual boot to windows 10.
> 
> Here is my hardware specs:

Hello. Are you using a System76 desktop, or some other desktop? We don't use the ZENITH II EXTREME in any of our machines to my knowledge, so I'm not sure the `system76-driver` repository is the correct repository for this issue, since `system76-driver` is meant to support System76 hardware.

Does the same issue occur if you boot a live disk of vanilla Ubuntu? I know we sometimes do work on the motherboards we use to get audio working correctly, and since we don't use that motherboard, it's possible nobody has done that work yet. That would likely need to be fixed in the Linux kernel itself by someone who is able to develop drivers for that motherboard.

If this is a System76 product, please open a [support case](https://support.system76.com/) so our support team can get some logs and figure out where the issue lies.

crawfxrd, 2022-07-22:
This is not System76 hardware.

closed 2022-07-22 by crawfxrd