pop-os/system76-driver #190: Arch /usr/share/initramfs-tools not found

elleven11, 2021-01-30:
**Distribution (run `cat /etc/os-release`):**
 NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://www.archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
aur/system76-driver 20.04.24-1 (+21 1.25)

**Issue/Bug Description:**
When i run the GUI software and the CLI one, as soon as i try to install drivers this error gets thrown:
```
INFO  product_version: 'oryp6'
INFO  blacklist_nvidia_i2c: Workaround for delay when loading NVIDIA i2c kernel module
INFO  Skipping 'blacklist_nvidia_i2c' as it was already applied
INFO  i915_initramfs: Add i915 driver to initramfs
INFO  Running 'i915_initramfs'
INFO  * Add i915 driver to initramfs
ERROR Error running actions:
Traceback (most recent call last):
  File "/usr/bin/system76-driver-cli", line 73, in <module>
    for msg in action_runner.run_iter():
  File "/usr/lib/python3.9/site-packages/system76driver/actions.py", line 253, in run_iter
    action.perform()
  File "/usr/lib/python3.9/site-packages/system76driver/actions.py", line 294, in perform
    self.atomic_write(self.content, self.mode)
  File "/usr/lib/python3.9/site-packages/system76driver/actions.py", line 215, in atomic_write
    fp = open(self.tmp, 'x')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/initramfs-tools/modules.d/s76-i915-initramfs.conf.5Z3OLVGR4TAM44ENBJAMZQPA'
```
Arch does not have the /usr/share/initramfs-tools folder at all.

I know that Arch is not officially supported, but I am required to use arch for my work.

**Steps to reproduce (if you know):**
I just installed system76-driver package from AUR and tried to install drivers.

**Expected behavior:**
Drivers would install.

yochananmarqos, 2021-02-23:
I added a [patch](https://aur.archlinux.org/cgit/aur.git/tree/actions.patch?h=system76-driver&id=fb495d68a608ba3274e302cba5242f37cce9ab71) that should fix it. Can you test `system76-driver` 20.04.28-3 and let me know if it works?

elleven11, 2021-02-24:
works. great.

closed 2021-02-24 by elleven11