pop-os/system76-driver #195: Please consider updating dependency on fwupd2 to newer version

davidkrauser, 2021-04-01:
**Distribution (run `cat /etc/os-release`):**
 
    NAME="KDE neon"
    VERSION="5.21"
    ID=neon
    ID_LIKE="ubuntu debian"
    PRETTY_NAME="KDE neon User Edition 5.21"
    VARIANT="User Edition"
    VARIANT_ID=user
    VERSION_ID="20.04"
    HOME_URL="https://neon.kde.org/"
    SUPPORT_URL="https://neon.kde.org/"
    BUG_REPORT_URL="https://bugs.kde.org/"
    LOGO=start-here-kde-neon
    PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
    VERSION_CODENAME=focal
    UBUNTU_CODENAME=focal
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
$ apt policy libfwupd2
libfwupd2:
  Installed: 1.4.5-1pop0~1611677200~20.04~aaada71~dev
  Candidate: 1.4.5-1pop0~1611677200~20.04~aaada71~dev
  Version table:
     1.5.7-0xneon+20.04+focal+release+build7 500
        500 http://archive.neon.kde.org/user focal/main amd64 Packages
     1.5.7-0xneon+20.04+focal+build6 500
        500 http://archive.neon.kde.org/user focal/main amd64 Packages
     1.5.6-0xneon+20.04+focal+build5 500
        500 http://archive.neon.kde.org/user focal/main amd64 Packages
     1.5.6-0xneon+20.04+focal+build4 500
        500 http://archive.neon.kde.org/user focal/main amd64 Packages
 *** 1.4.5-1pop0~1611677200~20.04~aaada71~dev 1001
       1001 http://ppa.launchpad.net/system76-dev/stable/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
     1.3.11-1~focal1 500
        500 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages
     1.3.9-4ubuntu0.1 500
        500 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages
     1.3.9-4 500
        500 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages
```

**Issue/Bug Description:**

With the latest version of KDE Neon User Edition, Discover (KDEs GUI Packaging Application) can't be installed:

    The following packages have unmet dependencies:
     plasma-discover : Depends: libfwupd2 (>= 1.5.7) but 1.4.5-1pop0~1611677200~20.04~aaada71~dev is to be installed

It depends on a newer version of libfwupd2 than what ships with the system76-driver package. Would you consider updating your dependency on libfwupd2 to help alleviate this issue?

**Steps to reproduce (if you know):**

- Install KDE Neon User Edition
- Execute:
```
sudo apt-add-repository -y ppa:system76-dev/stable
sudo apt update
sudo apt install system76-driver
```
- Note that Discover is uninstalled

**Expected behavior:**

Discover remains installed

jackpot51, 2021-04-01:
https://github.com/pop-os/fwupd/pull/4 should fix this

davidkrauser, 2021-04-01:
@jackpot51 awesome, thank you :smile:

davidkrauser, 2021-04-13:
This is now fixed. Thank you @jackpot51 :-)

closed 2021-04-13 by davidkrauser