pop-os/system76-driver #198: oryp7 speakers do not work (fix available in details)

MilesBHuff, 2021-04-30:
**Distribution (run `cat /etc/os-release`):**
 ```
NAME="Manjaro Linux"
ID=manjaro
ID_LIKE=arch
BUILD_ID=rolling
PRETTY_NAME="Manjaro Linux"
ANSI_COLOR="32;1;24;144;200"
HOME_URL="https://manjaro.org/"
DOCUMENTATION_URL="https://wiki.manjaro.org/"
SUPPORT_URL="https://manjaro.org/"
BUG_REPORT_URL="https://bugs.manjaro.org/"
LOGO=manjarolinux
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
`system76-driver 20.04.31-1`

**Issue/Bug Description:**
On oryp7, the speakers only work ootb on Pop!_OS.  OpenSUSE and Manjaro Linux are both completely mute after a fresh install (both OSes think that audio is working fine, but nothing comes out of the speakers).  Running `system76-driver-cli` does not resolve the issue.

**Other Notes:**
Adding the following to `/etc/modprobe.d/clevo-p950.conf` resolves the issue:
```
options snd-hda-intel model=clevo-p95
```

jacobgkau, 2021-04-30:
Rather than having a boot option set in the driver, Pop!_OS and our Ubuntu PPA have the proper configuration directly in the kernel: https://github.com/pop-os/linux/blob/ecb25cd0183314f9d7b729eed49bc8a153ab0b6e/sound/pci/hda/patch_realtek.c#L2555 

The other five System76 products listed in that file are already present in the upstream kernel, so I would expect we'll also attempt to upstream the oryp7 line. Once that happens, other distributions will eventually have working sound without needing the driver app installed.

ShadowAce, 2021-04-30:
This solution does not work under Fedora 34.  I just performed a fresh install of Fedora 34, added the above module (/etc/modprobe.d/clevo-p950.conf) and rebooted.  Still no sound.  snd-hda-intel was shown as loaded by alsa-info.sh

MilesBHuff, 2021-05-01:
@ShadowAce Interesting.  Maybe you also need `system76-acpi-dkms` or some other System76 software installed?

ShadowAce, 2021-05-03:
I have all the system76 packages installed from https://copr.fedorainfracloud.org/coprs/szydell/system76/

MilesBHuff, 2021-05-03:
@ShadowAce It looks like a few packages are missing from that;  and the instructions it gives are old -- they're for an oryp4.  I *just* finished writing the Arch Wiki article for the oryp7;  try following that and see if it works then?  You will have to build anything listed in the Arch Wiki but not listed in that Fedora repo from source, which seems to just be `system76-acpi-dkms` -- the same package I mentioned you needed in my first comment.  I also noticed that the instructions at that Fedora repo are missing `systemctl enable system76`;  try running that as well.

ShadowAce, 2021-05-03:
OK--Do you have a link for the Arch Wiki?

MilesBHuff, 2021-05-03:
@ShadowAce Yes;  I provided it in my first comment:  https://wiki.archlinux.org/title/System76_Oryx_Pro#Packages

szydell, 2021-05-04:
1. None of Fedora packages contain the 'system76.service', so enabling it won't work.
2. I just released system76-acpi-dkms (https://copr.fedorainfracloud.org/coprs/szydell/system76/package/system76-acpi-dkms/)

Blackrucat, 2021-07-15:
Install pipewire, worked for me.

MilesBHuff, 2021-07-26:
I was able to get it working most of the way with PulseAudio;  but PipeWire does work a bit better.

pilotstew, 2021-10-31:
I'm having this same issue with the newer oryx8 on arch. I have all system76 packages installed as recommended above and by the arch oryx pro wiki. I also added `options snd-hda-intel model=clevo-p950` to modprobe.d.

MilesBHuff, 2021-11-01:
@pilotstew 
> I also added `options snd-hda-intel model=clevo-p950` to modprobe.d.

It is likely that you are not on a clevo-p950, since you have a newer oryp.

> I have all system76 packages installed as recommended above and by the arch oryx pro wiki.

I'm actually the author of that wiki article;  but I wrote it for the oryp7, so it's likely that it is missing things that are needed for the oryp8.

@jacobgkau Do you have any advice for this situation?

jacobgkau, 2021-11-01:
> @jacobgkau Do you have any advice for this situation?

This is where oryp7's fixup used to be: https://github.com/pop-os/linux/blob/82fb22669ce05371a82414bc8569ad475ee2fba6/sound/pci/hda/patch_realtek.c#L2543 

In the current version, this has moved to an upstream Clevo line here (so oryp7 shouldn't need any options with recent kernels): https://github.com/pop-os/linux/blob/master/sound/pci/hda/patch_realtek.c 

The oryp8 is using the same `ALC1220_FIXUP_CLEVO_PB51ED_PINS` fixup as oryp7: https://github.com/pop-os/linux/blob/874b071285fe7b183cb510da864481aef9f25f4b/sound/pci/hda/patch_realtek.c#L2540 

Neither of these machines are a P950 model specifically, but the reason that worked on oryp7 is because the `clevo-p950` option applies ALC1220 fixups: https://www.kernel.org/doc/html/latest/sound/hd-audio/models.html#alc88x-898-1150-1220 

I have confirmed on a lab oryp8 that adding `snd-hda-intel.model=clevo-p950` to my boot options gets sound working in a Fedora 35 Beta live disk. @pilotstew I would try adding this at boot time within whatever bootloader your Arch install is using just to verify whether the issue is the option itself or getting it to apply.

jacobgkau, 2021-11-01:
I wanted to add for the record, one of the oryp8 fixup lines has already been upstreamed by another party, and @crawfxrd submitted the remaining fixup line upstream this morning: https://lore.kernel.org/all/20211101162134.5336-1-tcrawford@system76.com/ 

(The `model` option is still the best way to work around this until the upstream patches make their way back downstream into other distributions' kernels.)

pilotstew, 2021-11-01:
I can confirm the `options sna-hda-intel model=clevo-p950`  fix is working for the oryz8 on arch after fixing a typo. Thanks @MilesBHuff  & @jacobgkau

jacobgkau, 2021-11-02:
Closing this because oryp7/8 have been upstreamed and this was never an issue with the system76-driver app.

closed 2021-11-02 by jacobgkau