pop-os/system76-driver #213: No matter what I do kernel doesn't use GPU drivers

Blackrucat, 2021-07-15:
`

**Distribution (run `cat /etc/os-release`):**
```
 NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os
```

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
sudo apt policy system76-driver-nvidia
system76-driver-nvidia:
  Instalado: (nenhum)
  Candidato: 20.04.39~1626193589~21.04~d30d60b
  Tabela de versão:
     20.04.39~1626193589~21.04~d30d60b 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu hirsute/main amd64 Packages
       1001 http://ppa.launchpad.net/system76/pop/ubuntu hirsute/main i386 Packages
```

**Issue/Bug Description:**
Drivers don't work, nouveau, nvidia proprietary, none of them work, they have shown to be installed, but the kernel is not using them. So in consequence can't change my resolution, I can't use xrandr, nvidia-settings, I can't play minecraft  I am stucked at 10/20 fps, ON MINECRAFT with optifine!
```
**lspci -k**
00:00.0 Host bridge: Intel Corporation 82Q33 Express DRAM Controller (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82Q33 Express DRAM Controller
00:01.0 PCI bridge: Intel Corporation 82Q33 Express PCI Express Root Port (rev 02)
	Kernel driver in use: pcieport
00:03.0 Communication controller: Intel Corporation 82Q33 Express MEI Controller (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82Q33 Express MEI Controller
	Kernel driver in use: mei_me
	Kernel modules: mei_me
00:1a.0 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #4 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB UHCI Controller
	Kernel driver in use: uhci_hcd
00:1a.1 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #5 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB UHCI Controller
	Kernel driver in use: uhci_hcd
00:1a.2 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #6 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB UHCI Controller
	Kernel driver in use: uhci_hcd
00:1a.7 USB controller: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #2 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB2 EHCI Controller
	Kernel driver in use: ehci-pci
00:1b.0 Audio device: Intel Corporation 82801I (ICH9 Family) HD Audio Controller (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) HD Audio Controller
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
00:1c.0 PCI bridge: Intel Corporation 82801I (ICH9 Family) PCI Express Port 1 (rev 02)
	Kernel driver in use: pcieport
00:1c.4 PCI bridge: Intel Corporation 82801I (ICH9 Family) PCI Express Port 5 (rev 02)
	Kernel driver in use: pcieport
00:1c.5 PCI bridge: Intel Corporation 82801I (ICH9 Family) PCI Express Port 6 (rev 02)
	Kernel driver in use: pcieport
00:1d.0 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #1 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB UHCI Controller
	Kernel driver in use: uhci_hcd
00:1d.1 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #2 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB UHCI Controller
	Kernel driver in use: uhci_hcd
00:1d.2 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #3 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB UHCI Controller
	Kernel driver in use: uhci_hcd
00:1d.7 USB controller: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #1 (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) USB2 EHCI Controller
	Kernel driver in use: ehci-pci
00:1e.0 PCI bridge: Intel Corporation 82801 PCI Bridge (rev 92)
00:1f.0 ISA bridge: Intel Corporation 82801IB (ICH9) LPC Interface Controller (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801IB (ICH9) LPC Interface Controller
	Kernel driver in use: lpc_ich
	Kernel modules: lpc_ich
00:1f.2 IDE interface: Intel Corporation 82801IB (ICH9) 2 port SATA Controller [IDE mode] (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801IB (ICH9) 2 port SATA Controller [IDE mode]
	Kernel driver in use: ata_piix
	Kernel modules: pata_acpi
00:1f.3 SMBus: Intel Corporation 82801I (ICH9 Family) SMBus Controller (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) SMBus Controller
	Kernel driver in use: i801_smbus
	Kernel modules: i2c_i801
00:1f.5 IDE interface: Intel Corporation 82801I (ICH9 Family) 2 port SATA Controller [IDE mode] (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc 82801I (ICH9 Family) 2 port SATA Controller [IDE mode]
	Kernel driver in use: ata_piix
	Kernel modules: pata_acpi
01:00.0 VGA compatible controller: NVIDIA Corporation G72 [GeForce 7200 GS / 7300 SE] (rev a1)
	Subsystem: ZOTAC International (MCO) Ltd. G72 [GeForce 7200 GS / 7300 SE]
	Kernel modules: nvidiafb, nouveau, nvidia_drm, nvidia
03:00.0 IDE interface: JMicron Technology Corp. JMB368 IDE controller
	Subsystem: FIRST INTERNATIONAL Computer Inc JMB368 IDE controller
	Kernel driver in use: pata_jmicron
	Kernel modules: pata_jmicron, pata_acpi
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 02)
	Subsystem: FIRST INTERNATIONAL Computer Inc RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
	Kernel driver in use: r8169
	Kernel modules: r8169
```
```
**lspci -k | grep -A 2 -i "VGA"**
01:00.0 VGA compatible controller: NVIDIA Corporation G72 [GeForce 7200 GS / 7300 SE] (rev a1)
	Subsystem: ZOTAC International (MCO) Ltd. G72 [GeForce 7200 GS / 7300 SE]
	Kernel modules: nvidiafb, nouveau, nvidia_drm, nvidia
```
```
**glxinfo | egrep "OpenGL vendor|OpenGL renderer"**
OpenGL vendor string: Mesa/X.org
OpenGL renderer string: llvmpipe (LLVM 11.0.1, 128 bits)
```

**Steps to reproduce (if you know):**
Install Pop Os on computer, perceive that video card isn't working well, struggle trying to fix the resolution, try many things, use the command to check if the driver is installed, but it gives a  r8169 instead, sit and cry, do a commit and cry.
Sorry, I can't be serious, but I've spend a lot of time thinking this was my problem and I tried to fix that for months, but I noticed that was much better to do a commit.

**Expected behavior:**
I know the proprietary driver isn't appropriated for any old video card, but I need help to get nouveau drivers working, I just wanted to play some games, and do my school stuff, it's really hard to do in the tiny resolution that I am using by now. I want to change my resolution and play games, just like I would do in Windows, I would get 180(MAX) fps on Minecraft.

**Other Notes:**

leviport, 2021-07-15:
Looking on the NVIDIA site, it looks like the latest driver they have listed that supported the 7200 was 304: https://www.nvidia.com/Download/driverResults.aspx/123708/en-us

This isn't in any repos, so you'd have to install it manually. It looks like that GPU [was released in 2006](https://www.techpowerup.com/gpu-specs/geforce-7200-gs.c723) so its age might make it difficult to do a whole lot of gaming with it.

closed 2021-07-15 by leviport

Blackrucat, 2021-07-15:
Doesn't nouveau support it?

leviport, 2021-07-15:
I can't say for certain, but Nouveau's first release was 6 years after that GPU was released. My searching turned up this stack exchange thread, and it looks like the poster didn't have much luck switching to Nouveau: https://unix.stackexchange.com/questions/120970/switching-from-nvidia-proprietary-driver-to-nouveau-driver-for-nvidia-card-on-de

Manually installing the 304 driver might be your best bet.

Blackrucat, 2021-07-18:
Everytime I try to install  the nvidia 304 .run it doesn't work, even on tty, disabling gdm or whatever.
IDK if it's because of the kernel, but it doesn't work.

leviport, 2021-07-19:
Those .run files definitely take a lot of fiddling. I've regularly had better luck installing them on Ubuntu instead of Pop since it sometimes doesn't like some of the newer kernels Pop usually has. Also, Ubuntu might have more driver options in the repos. It might be worth trying out Ubuntu. Then, if you can make a configuration work in Ubuntu (and still feel like fiddling with it more afterwards), you might be able to replicate the same driver/configuration on a Pop install.

Blackrucat, 2021-08-11:
I am using ZorinOS right now, for some reason Xrandr works properly and I can change my resolution, even using the latest kernel version. The only distro that I've found problem with nouveau was Pop!_OS! I am no more trying to use the proprietary drivers, because is making me sad I am just loosing my time. Anyways thank you guys for trying to help me, and please fix the nouveau and xrandr problem on Pop!_OS! It happens only with legacy videocards like mine.

Blackrucat, 2021-08-19:
Nouveau don't work properly on Pop!_OS! And works in 100% of the distros that I've tested.
I could change my resolution, easily, do a ./xprofile and use xrandr without gama error in every other distro, but POP! os refuses to do that!

leviport, 2021-08-19:
>without gama error

Can you post this error? I don't think I've seen that one before.

Blackrucat, 2021-08-19:
I am not using Pop!_OS! anymore, but I can give you an example that's exactly the same thing
[image]

It's just like that.
(I took this example from reddit.)