pop-os/system76-driver #216: gaze16 headset jack not working in Arch Linux

JCDetwiler, 2021-09-29:
**Distribution:** Arch Linux
 
**Related Application and/or Package Version:**
* `pulseaudio 15.0-1`
* `pavucontrol 1:5.0-1`

**Issue/Bug Description:**

My headset can only play output audio and the mic doesn't work; this is on a gaze16 laptop running Arch Linux.

**Steps to reproduce (if you know):**

When I plug in my Jabra headset, the whole headset isn't detected in any fashion. I'm able to switch to `Headphones (unplugged)` in `pavucontrol` as an output device, but I can't do the same for the mic.

**Expected behavior:**

The headset and its mic should both work when plugged into the main audio jack directly.

**Other Notes:**

I have all of the necessary firmware/drivers from [here][1] installed. Sorry if `system76-driver` isn't the right place for this issue.

[1]: https://support.system76.com/articles/system76-software

crawfxrd, 2022-07-22:
Is this still reproducible?

Headset detection is enabled by a kernel quirk, and at this point is in all stable kernels.

JCDetwiler, 2022-10-19:
Sorry for the delay, I think it has since been fixed.

closed 2022-10-19 by JCDetwiler