pop-os/system76-driver #309: System76 Module Build Error on NixOS with Kernel 6.12

kylegortych, 2024-11-26:
Apologies if this is a duplicate or more specifically a nix related issue. I'm encountering an error when running ```sudo nixos-rebuild boot --upgrade ``` it fails with an error related to the System76 module which seems to be related to building with latest Linux kernel version.

```plaintext
"Specifically, the error indicates an incompatible pointer type being passed to
the wmi_install_notify_handler function, which suggests the System76 driver needs
to be updated to match the current kernel API." 

Source: Perplexity AI, accessed on 11/26/2024
```

Thank you in advance for any help and assistance.

<details open>
<summary>stderr</summary>

```shell
sudo nixos-rebuild boot --upgrade --show-trace
unpacking 2 channels...
building Nix...
building the system configuration...
these 29 derivations will be built:
  /nix/store/c33hg07wjy2xh3icrkjqybcfbc4clbk9-system-path.drv
  /nix/store/ic0hq3w40q28vm9jyxvccsapk07d5wxl-dbus-1.drv
  /nix/store/4p2fd90p30784i59z61h9xsm2lr2mykj-X-Restart-Triggers-dbus.drv
  /nix/store/pwnyr2cdqm29cwj9zfdz2kv54vc7zp06-unit-dbus.service.drv
  /nix/store/44zg0wgp0y562bgqc8qnfqgnr4kpaln6-user-units.drv
  /nix/store/25n1l11xjssq37xp3j08xnhlvhd1drfp-unit-accounts-daemon.service.drv
  /nix/store/vm4nfvzn4v74jbzc3ssnpw9g31py2kv2-google-chrome-131.0.6778.69.drv
  /nix/store/y3q78iwym0x55azqpr0vzsaddmds2hgd-neovim-0.10.2.drv
  /nix/store/7cjvk1p1d46ar3jsa22m7w6ipnfsbz7f-home-manager-path.drv
  /nix/store/iksrw4l5gympja1zi7jcgg42b6qxcg5x-activation-script.drv
  /nix/store/k0nnxsjmc775qrbbd247yf7ylz9gc583-home-manager-generation.drv
  /nix/store/3qws34fpa4kjwn2vnhjgf6qchiq54df7-unit-home-manager-kg.service.drv
  /nix/store/8lxnlx48dg4d1hibm3vf3v5i92gmv7lb-unit-dbus.service.drv
  /nix/store/8rjkrhyjnrhiqsvvbvy0rmcrh9ihd3xi-X-Restart-Triggers-polkit.drv
  /nix/store/rxgnfcmzjvh14amd4g60y0lif8bgjc65-unit-polkit.service.drv
  /nix/store/d2n0lh0z8y1wyk11iwwhy9xvjcbbz6lp-system76-module-1.0.16-6.12.drv
  /nix/store/k5nrz09qhny9hx5l9qa6wj3j09l8ncra-udev-rules.drv
  /nix/store/m5i4mzxf7ygngg3rmk6r9x8rgafqrr5p-X-Restart-Triggers-systemd-udevd.drv
  /nix/store/x739z4kx4ky2pzrsp0ikk6vk392mh9yi-unit-systemd-udevd.service.drv
  /nix/store/hnrkkjck1fcyrh3d936aiz2yknr3l9k3-system-units.drv
  /nix/store/shk1nvs23120l11zbpba60qjhrb1pbb7-hwdb.bin.drv
  /nix/store/4746axv504k0cl5nxr6j17jcmn8ik41v-man-cache.drv
  /nix/store/yb3iri9acdgfpwcw6z4l26x8yz5dimnx-etc-man_db.conf.drv
  /nix/store/fzc4lw9b95hgqhg58k9xhvaxr95rb2if-etc.drv
  /nix/store/pydc109m5b0bamwsim47cll1mzxbvjyh-linux-6.12-modules.drv
  /nix/store/glpcz3nh0hnvqh1h51f6na08b4bmz6fq-linux-6.12-modules-shrunk.drv
  /nix/store/s6xx6fldhqhrv7q1byra1w76dasgvigg-initrd-linux-6.12.drv
  /nix/store/ql9l7x09zc9fw8fpyppphn3i09psx1r0-boot.json.drv
  /nix/store/1q9h71a6ax5i5wsf86d344wxvk22z2p9-nixos-system-dev-null-25.05beta710087.23e89b7da85c.drv
building '/nix/store/c33hg07wjy2xh3icrkjqybcfbc4clbk9-system-path.drv'...
building '/nix/store/d2n0lh0z8y1wyk11iwwhy9xvjcbbz6lp-system76-module-1.0.16-6.12.drv'...
building '/nix/store/4746axv504k0cl5nxr6j17jcmn8ik41v-man-cache.drv'...
building '/nix/store/vm4nfvzn4v74jbzc3ssnpw9g31py2kv2-google-chrome-131.0.6778.69.drv'...
building '/nix/store/y3q78iwym0x55azqpr0vzsaddmds2hgd-neovim-0.10.2.drv'...
Running phase: unpackPhase
unpacking source archive /nix/store/hzshnr6nhpdiz6rgqabsp620r5iqlw0h-source
source root is source
Running phase: patchPhase
Running phase: updateAutotoolsGnuConfigScriptsPhase
Running phase: configurePhase
no configure script, doing nothing
Running phase: buildPhase
build flags: SHELL=/nix/store/0irlcqx2n3qm6b1pc9rsd2i8qpvcccaj-bash-5.2p37/bin/bash KERNEL_DIR=/nix/store/lacdcr5zi7w59f8xsq4cfi41mzgc6dcn-linux-6.12-dev/lib/modules/6.12.0/build
make -C "/nix/store/lacdcr5zi7w59f8xsq4cfi41mzgc6dcn-linux-6.12-dev/lib/modules/6.12.0/build" M="/build/source" modules
make[1]: Entering directory '/nix/store/lacdcr5zi7w59f8xsq4cfi41mzgc6dcn-linux-6.12-dev/lib/modules/6.12.0/build'
structuredAttrs is enabled
Running phase: patchPhase
Running phase: updateAutotoolsGnuConfigScriptsPhase
Running phase: configurePhase
[... 85 lines cut ...]
  sddmwallpaper = /home/blank/Pictures/wallpaper.jpg;
in
{
  imports =
    [ # Include the results of the hardware scan.
      ./hardware-configuration.nix
      <home-manager/nixos>
      ./dotfiles/doom.nix
      ./dotfiles/fish.nix
      ./dotfiles/neovim.nix
      ./dotfiles/starship.nix
      ./dotfiles/wezterm.nix
    ];

  #nix settings
  nix.settings.experimental-features = [ "nix-command" "flakes" ];

  hardware.system76.enableAll = true;
  
  ... etc.  

}
```

</details>

ahoneybun, 2024-12-02:
Does it work if you pull in the changes from nixos-hardware? It works on my galp5-1650

https://gitlab.com/ahoneybun/nix-configs/-/blob/main/flake.nix#L83

kylegortych, 2024-12-02:
Hello @ahoneybun, I'm not sure if I can pull the changes as I haven't used flakes yet. I did run ```sudo nix-channel --update```. This problem has happen before and it was resolved so it may just be a matter of waiting. 

Thank you for the link to your flake config. I may need to start using flakes.

kylegortych, 2024-12-13:
It seems my laptop model hasn't been added yet.  

Also, do I still use ```hardware.system76.enableAll = true;``` if using the nixos-hardware channel ```<nixos-hardware/system76/pangolin/>``` if it is added?

@ahoneybun

ahoneybun, 2024-12-13:
If that model is in nixos-hardware it would already include that option and it's value.

kylegortych, 2024-12-14:
Currently commented out ```hardware.system76.enableAll = true;``` and rebuild successfully to latest kernel version. Everything is working which might be do to hardware scan from live-iso install. 

Thank you for your help and assistance.

closed 2024-12-14 by kylegortych

ahoneybun, 2024-12-16:
The issue is actually this:

https://github.com/pop-os/system76-dkms/pull/71

It has been merged now! Now the following needs to happen:

1. A new release upstream
2. A new PR for NixOS based on that