pop-os/system76-driver #325: Possible System76 and Linux kernel conflict

Jedburgh911, 2025-02-26:
Hit:1 https://download.docker.com/linux/ubuntu noble InRelease
Hit:2 https://updates.signal.org/desktop/apt xenial InRelease
Hit:3 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:4 https://repo.ivpn.net/stable/ubuntu ./generic InRelease
Hit:5 http://archive.ubuntu.com/ubuntu noble InRelease
Hit:6 https://packages.microsoft.com/repos/code stable InRelease
Hit:7 http://security.ubuntu.com/ubuntu noble-security InRelease
Hit:8 http://deb.anydesk.com all InRelease
Hit:9 https://deb.packages.mattermost.com stable InRelease
Hit:10 https://deb.oxen.io noble InRelease
Hit:11 http://archive.ubuntu.com/ubuntu noble-updates InRelease
Hit:12 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Hit:13 https://packages.cloud.google.com/apt cloud-sdk InRelease
Hit:14 https://esm.ubuntu.com/apps/ubuntu noble-apps-security InRelease
Hit:15 https://esm.ubuntu.com/apps/ubuntu noble-apps-updates InRelease
Hit:17 https://esm.ubuntu.com/infra/ubuntu noble-infra-security InRelease
Hit:18 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu noble InRelease
Hit:19 https://esm.ubuntu.com/infra/ubuntu noble-infra-updates InRelease
Hit:20 https://ppa.launchpadcontent.net/mozillateam/ppa/ubuntu noble InRelease
Hit:21 https://ppa.launchpadcontent.net/system76-dev/stable/ubuntu noble InRelease
Hit:16 https://ubuntu.qgis.org/ubuntu noble InRelease
Reading package lists...
Building dependency tree...
Reading state information...
All packages are up to date.
W: https://ppa.launchpadcontent.net/system76-dev/stable/ubuntu/dists/noble/InRelease: Signature by key 5D1F3A80254F6AFBA254FED5ACD442D1C8B7748B uses weak algorithm (rsa1024)

Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages were automatically installed and are no longer required:
  amd-ppt-bin amd64-microcode at dfu-programmer dkms firmware-manager
  firmware-manager-notify firmware-manager-shared
  gnome-shell-extension-system76-power hidpi-daemon libnvidia-cfg1-560
  libnvidia-common-560 libnvidia-decode-560 libnvidia-egl-wayland1
  libnvidia-encode-560 libnvidia-extra-560 libnvidia-fbc1-560 libnvidia-gl-560
  libpkgconf3 libxnvctrl0 linux-headers-6.12.10-76061203
  linux-headers-6.12.10-76061203-generic linux-headers-generic
  linux-modules-6.12.10-76061203-generic nvidia-compute-utils-560
  nvidia-dkms-560 nvidia-driver-550 nvidia-driver-560
  nvidia-firmware-560-560.35.03 nvidia-kernel-common-560
  nvidia-kernel-source-560 nvidia-settings nvidia-utils-560 pkg-config pkgconf
  pkgconf-bin python3-evdev python3-pydbus screen-resolution-extra
  system76-acpi-dkms system76-dkms system76-firmware-daemon system76-io-dkms
  system76-oled system76-wallpapers thermald xbacklight
  xserver-xorg-video-nvidia-560
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Setting up linux-headers-6.12.10-76061203-generic (6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev) ...
/etc/kernel/header_postinst.d/dkms:
 * dkms: running auto installation service for kernel 6.12.10-76061203-generic
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
unset ARCH; [ ! -h /usr/bin/cc ] && export CC=/usr/bin/gcc; env NV_VERBOSE=1 'make' -j16 NV_EXCLUDE_BUILD_MODULES='' KERNEL_UNAME=6.12.10-76061203-generic IGNORE_XEN_PRESENCE=1 IGNORE_CC_MISMATCH=1 SYSSRC=/lib/modules/6.12.10-76061203-generic/build LD=/usr/bin/ld.bfd CONFIG_X86_KERNEL_IBT= modules.........(bad exit status: 2)
ERROR (dkms apport): kernel package linux-headers-6.12.10-76061203-generic is not supported
Error! Bad return status for module build on kernel: 6.12.10-76061203-generic (x86_64)
Consult /var/lib/dkms/nvidia/560.35.03/build/make.log for more information.
dkms autoinstall on 6.12.10-76061203-generic/x86_64 succeeded for system76 system76_acpi system76-io
dkms autoinstall on 6.12.10-76061203-generic/x86_64 failed for nvidia(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 * dkms: autoinstall for kernel 6.12.10-76061203-generic
   ...fail!
run-parts: /etc/kernel/header_postinst.d/dkms exited with return code 11
dpkg: error processing package linux-headers-6.12.10-76061203-generic (--configure):
 installed linux-headers-6.12.10-76061203-generic package post-installation script subprocess returned error exit status 11
dpkg: dependency problems prevent configuration of linux-headers-generic:
 linux-headers-generic depends on linux-headers-6.12.10-76061203-generic (= 6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev); however:
  Package linux-headers-6.12.10-76061203-generic is not configured yet.

dpkg: error processing package linux-headers-generic (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 linux-headers-6.12.10-76061203-generic
 linux-headers-generic

Jedburgh911, 2025-02-26:
Continue to received "weak algorithm" notification at apt update.

harrybrwn, 2025-02-27:
I'm also having the same issue. I may have mistakenly posted in the wrong place but [my logs are over here](https://github.com/pop-os/nvidia-graphics-drivers-565/issues/1).

harrybrwn, 2025-02-27:
@Jedburgh911 Are there any warnings or errors that stick out in `/var/lib/dkms/nvidia/560.35.03/build/make.log`?

jackpot51, 2025-02-27:
I think this will be fixed by updating the system76-driver-nvidia package to version 22.04.1 or 24.04.2, depending on your OS version.

harrybrwn, 2025-02-27:
system76-driver-nvidia is what I'm trying to update, the issue looks like its because, when upgrading it's dependencies, `dpkg --configure linux-headers-6.12.10-76061203-generic` and `dpkg --configure linux-image-6.12.10-76061203-generic` are both triggering a dkms build which is failing. I could still have it wrong but the above commands are triggering the same errors. 

This is what my `dpkg --list` looks like:
```
ii  linux-firmware                                    20240610.git1d67e66d-0pop1+system76~1732201256~24.04~9061b5f~dev      amd64
iU  linux-generic                                     6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev            amd64
ii  linux-headers-6.12.10-76061203                    6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev            all
iF  linux-headers-6.12.10-76061203-generic            6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev            amd64
ii  linux-headers-6.9.3-76060903                      6.9.3-76060903.202405300957~1738771449~24.04~b48c952~dev              all
ii  linux-headers-6.9.3-76060903-generic              6.9.3-76060903.202405300957~1738771449~24.04~b48c952~dev              amd64
iU  linux-headers-generic                             6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev            amd64
iF  linux-image-6.12.10-76061203-generic              6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev            amd64
rc  linux-image-6.8.0-31-generic                      6.8.0-31.31                                                           amd64
iU  linux-system76                                    6.12.10-76061203.202412060638~1740098661~24.04~f50b0de~dev            amd64
iU  system76-driver                                   24.04.2~1740604629~24.04~8c83d89~dev                                  amd64
iU  system76-driver-nvidia                            24.04.2~1740604629~24.04~8c83d89~dev                                  amd64
```
It looks like they are stuck in "Unpack" because linux-image and linux-headers are failing.

harrybrwn, 2025-02-27:
Oh, do I have the wrong driver installed? I can see that being the issue.

```
  ii  nvidia-compute-utils-560                          560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  ii  nvidia-dkms-560                                   560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  ii  nvidia-driver-550                                 560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  ii  nvidia-driver-560                                 560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  ii  nvidia-firmware-560-560.35.03                     560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  rc  nvidia-kernel-common-550:amd64                    560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  ii  nvidia-kernel-common-560                          560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  ii  nvidia-kernel-source-560                          560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
  ii  nvidia-settings                                   570.86.10-0ubuntu1                                                    amd64
  ii  nvidia-utils-560                                  560.35.03-1pop0~1729800494~24.04~4a3cad6~dev                          amd64
```

jackpot51, 2025-02-27:
Our packaging should be able to transition you from 560 to 565. Are you able to install nvidia-driver-565?

harrybrwn, 2025-02-27:
I'm getting:
```
 $ sudo apt install nvidia-driver-565
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package nvidia-driver-565
```

I found a [reddit thread](https://www.reddit.com/r/linux_gaming/comments/1hi7lf4/nvidia_56577_drivers_available_if_you_use_ubuntu/) that says to add the "ppa:graphics-drivers/ppa" ppa, am I supposed to have that on my machine already?

harrybrwn, 2025-02-27:
I also remember seeing a list of nvidia drivers here at some point. I'm not sure what I did in the last few weeks to cause this.

[image]

harrybrwn, 2025-02-27:
Yes.

```
harry@zephyr:/etc/apt/sources.list.d $ cat /etc/apt/sources.list.d/system76-dev-ubuntu-stable-noble.sources
Types: deb
URIs: https://ppa.launchpadcontent.net/system76-dev/stable/ubuntu/
Suites: noble
Components: main
Signed-By: -----BEGIN PGP PUBLIC KEY BLOCK-----
...
-----END PGP PUBLIC KEY BLOCK-----
```

However I regularly get the warning. I'm not sure if that suggests that its the wrong one?

```
W: https://ppa.launchpadcontent.net/system76-dev/stable/ubuntu/dists/noble/InRelease: Signature by key 5D1F3A80254F6AFBA254FED5ACD442D1C8B7748B uses weak algorithm (rsa1024)
```

jackpot51, 2025-02-27:
I'm having our QA team look at this, hopefully we can update you soon

harrybrwn, 2025-02-27:
Awesome, thanks for looking into this.

PSA: for anyone else running into this issue run this before **shutting down** or **rebooting**:

```
sudo update-initramfs -u -k $(uname -r)
sudo update-grub
```

I had problems booting with out it.

Jedburgh911, 2025-02-27:
Sorry guys been on the road for the last day or so. Running update now and it looks for an updated System76, so far  so good, but here's yesterday's (EST) compile log:

`DKMS` make.log for nvidia-560.35.03 for kernel 6.12.10-76061203-generic (x86_64)
Wed Feb 26 05:44:58 PM EST 2025
make[1]: Entering directory '/usr/src/linux-headers-6.12.10-76061203-generic'
make --no-print-directory -C /usr/src/linux-headers-6.12.10-76061203-generic \
-f /usr/src/linux-headers-6.12.10-76061203-generic/Makefile modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-14 (Ubuntu 14.2.0-4ubuntu2~24.04) 14.2.0
  You are using:           cc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
make -f ./scripts/Makefile.build obj=/var/lib/dkms/nvidia/560.35.03/build need-builtin=1 need-modorder=1 
# cmd_mod /var/lib/dkms/nvidia/560.35.03/build/nvidia.mod
  printf '%s
'   nvidia/nv.o nvidia/nv-pci.o nvidia/nv-dmabuf.o nvidia/nv-nano-timer.o nvidia/nv-acpi.o nvidia/nv-cray.o nvidia/nv-dma.o nvidia/nv-i2c.o nvidia/nv-mmap.o nvidia/nv-p2p.o nvidia/nv-pat.o nvidia/nv-procfs.o nvidia/nv-usermap.o nvidia/nv-vm.o nvidia/nv-vtophys.o nvidia/os-interface.o nvidia/os-mlock.o nvidia/os-pci.o nvidia/os-registry.o nvidia/os-usermap.o nvidia/nv-modeset-interface.o nvidia/nv-pci-table.o nvidia/nv-kthread-q.o nvidia/nv-memdbg.o nvidia/nv-ibmnpu.o nvidia/nv-report-err.o nvidia/nv-rsync.o nvidia/nv-msi.o nvidia/nv-caps.o nvidia/nv-caps-imex.o nvidia/nv-host1x.o nvidia/nv_uvm_interface.o nvidia/libspdm_aead.o nvidia/libspdm_ecc.o nvidia/libspdm_hkdf.o nvidia/libspdm_rand.o nvidia/libspdm_shash.o nvidia/libspdm_rsa.o nvidia/libspdm_aead_aes_gcm.o nvidia/libspdm_sha.o nvidia/libspdm_hmac_sha.o nvidia/libspdm_internal_crypt_lib.o nvidia/libspdm_hkdf_sha.o nvidia/libspdm_ec.o nvidia/libspdm_x509.o nvidia/libspdm_rsa_ext.o nvidia/nvlink_linux.o nvidia/nvlink_caps.o nvidia/linux_nvswitch.o nvidia/procfs_nvswitch.o nvidia/i2c_nvswitch.o nvidia/nv-kernel.o | awk '!x[$0]++ { print("/var/lib/dkms/nvidia/560.35.03/build/"$0) }' > /var/lib/dkms/nvidia/560.35.03/build/nvidia.mod
# SYMLINK /var/lib/dkms/nvidia/560.35.03/build/nvidia/nv-kernel.o
  ln -sf /var/lib/dkms/nvidia/560.35.03/build/nvidia/nv-kernel.o_binary /var/lib/dkms/nvidia/560.35.03/build/nvidia/nv-kernel.o
# cmd_mod /var/lib/dkms/nvidia/560.35.03/build/nvidia-uvm.mod
  printf '%s
'   nvidia-uvm/uvm_ats_sva.o nvidia-uvm/uvm_conf_computing.o nvidia-uvm/uvm_sec2_test.o nvidia-uvm/uvm_maxwell_sec2.o nvidia-uvm/uvm_hopper_sec2.o nvidia-uvm/uvm_blackwell.o nvidia-uvm/uvm_blackwell_fault_buffer.o nvidia-uvm/uvm_blackwell_mmu.o nvidia-uvm/uvm_blackwell_host.o nvidia-uvm/uvm_common.o nvidia-uvm/uvm_linux.o nvidia-uvm/nvstatus.o nvidia-uvm/nvCpuUuid.o nvidia-uvm/nv-kthread-q.o nvidia-uvm/nv-kthread-q-selftest.o nvidia-uvm/uvm.o nvidia-uvm/uvm_tools.o nvidia-uvm/uvm_global.o nvidia-uvm/uvm_gpu.o nvidia-uvm/uvm_gpu_isr.o nvidia-uvm/uvm_procfs.o nvidia-uvm/uvm_va_space.o nvidia-uvm/uvm_va_space_mm.o nvidia-uvm/uvm_gpu_semaphore.o nvidia-uvm/uvm_mem.o nvidia-uvm/uvm_rm_mem.o nvidia-uvm/uvm_channel.o nvidia-uvm/uvm_lock.o nvidia-uvm/uvm_hal.o nvidia-uvm/uvm_processors.o nvidia-uvm/uvm_range_tree.o nvidia-uvm/uvm_rb_tree.o nvidia-uvm/uvm_range_allocator.o nvidia-uvm/uvm_va_range.o nvidia-uvm/uvm_va_policy.o nvidia-uvm/uvm_va_block.o nvidia-uvm/uvm_range_group.o nvidia-uvm/uvm_gpu_replayable_faults.o nvidia-uvm/uvm_gpu_non_replayable_faults.o nvidia-uvm/uvm_gpu_access_counters.o nvidia-uvm/uvm_perf_events.o nvidia-uvm/uvm_perf_module.o nvidia-uvm/uvm_mmu.o nvidia-uvm/uvm_pte_batch.o nvidia-uvm/uvm_tlb_batch.o nvidia-uvm/uvm_push.o nvidia-uvm/uvm_pushbuffer.o nvidia-uvm/uvm_thread_context.o nvidia-uvm/uvm_tracker.o nvidia-uvm/uvm_maxwell.o nvidia-uvm/uvm_maxwell_host.o nvidia-uvm/uvm_maxwell_ce.o nvidia-uvm/uvm_maxwell_mmu.o nvidia-uvm/uvm_maxwell_fault_buffer.o nvidia-uvm/uvm_maxwell_access_counter_buffer.o nvidia-uvm/uvm_pascal.o nvidia-uvm/uvm_pascal_ce.o nvidia-uvm/uvm_pascal_host.o nvidia-uvm/uvm_pascal_mmu.o nvidia-uvm/uvm_pascal_fault_buffer.o nvidia-uvm/uvm_volta_ce.o nvidia-uvm/uvm_volta_host.o nvidia-uvm/uvm_volta_mmu.o nvidia-uvm/uvm_volta.o nvidia-uvm/uvm_volta_fault_buffer.o nvidia-uvm/uvm_volta_access_counter_buffer.o nvidia-uvm/uvm_turing.o nvidia-uvm/uvm_turing_access_counter_buffer.o nvidia-uvm/uvm_turing_fault_buffer.o nvidia-uvm/uvm_turing_mmu.o nvidia-uvm/uvm_turing_host.o nvidia-uvm/uvm_ampere.o nvidia-uvm/uvm_ampere_ce.o nvidia-uvm/uvm_ampere_host.o nvidia-uvm/uvm_ampere_fault_buffer.o nvidia-uvm/uvm_ampere_mmu.o nvidia-uvm/uvm_hopper.o nvidia-uvm/uvm_hopper_fault_buffer.o nvidia-uvm/uvm_hopper_ce.o nvidia-uvm/uvm_hopper_host.o nvidia-uvm/uvm_hopper_mmu.o nvidia-uvm/uvm_ada.o nvidia-uvm/uvm_policy.o nvidia-uvm/uvm_perf_utils.o nvidia-uvm/uvm_kvmalloc.o nvidia-uvm/uvm_pmm_sysmem.o nvidia-uvm/uvm_pmm_gpu.o nvidia-uvm/uvm_migrate.o nvidia-uvm/uvm_populate_pageable.o nvidia-uvm/uvm_migrate_pageable.o nvidia-uvm/uvm_map_external.o nvidia-uvm/uvm_user_channel.o nvidia-uvm/uvm_hmm.o nvidia-uvm/uvm_perf_heuristics.o nvidia-uvm/uvm_perf_thrashing.o nvidia-uvm/uvm_perf_prefetch.o nvidia-uvm/uvm_ats.o nvidia-uvm/uvm_ats_ibm.o nvidia-uvm/uvm_ats_faults.o nvidia-uvm/uvm_test.o nvidia-uvm/uvm_test_rng.o nvidia-uvm/uvm_range_tree_test.o nvidia-uvm/uvm_range_allocator_test.o nvidia-uvm/uvm_gpu_semaphore_test.o nvidia-uvm/uvm_mem_test.o nvidia-uvm/uvm_rm_mem_test.o nvidia-uvm/uvm_page_tree_test.o nvidia-uvm/uvm_tracker_test.o nvidia-uvm/uvm_push_test [... cut ...]
cc: error: unrecognized command-line option ‘-fmin-function-alignment=16’; did you mean ‘-flimit-function-alignment’?
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/nvidia/560.35.03/build/nvidia/nv-mmap.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.10-76061203-generic/Makefile:1943: /var/lib/dkms/nvidia/560.35.03/build] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.10-76061203-generic'
make: *** [Makefile:89: modules] Error 2`

jackpot51, 2025-02-27:
We should have updates available that fix this soon, we are just waiting on builds to finish.

harrybrwn, 2025-02-28:
Ok I ran `apt update` and `apt install nvidia-driver-565` and rebooted and nothing has failed yet 🤞.

jackpot51, 2025-02-28:
It should be fixed now. Our PPA ran out of space and that package was not uploaded as expected.

Jedburgh911, 2025-03-03:
This issue seems to be resolved after the latest push. I have run `apt update`  and `apt upgrade` successfully and my GPU is integrated into system functions successfully. Thank you.

closed 2025-03-03 by Jedburgh911